Protonerd/FX-SaberOS #66: Occasionnally hum keeps playing after turning the saber off

DonSaber, 2018-03-05:
This is the second occasional problem I noticed while using extensively the saber to a sparring class.

Sometimes it happens that when you disarm the saber, the lights go off but the hum keeps playing.
The only way to stop the hum seems to be plugging the kill plug to cut the power to the board.
Also this problem is intermittent and most of the times the saber shuts down just fine.

I'm running previous FX-SaberOS build with no custom mods, apart from H/W config and MPU calibration values.

DonSaber, 2018-03-05:
Forgot to mention that I'm using Tri-Cree leds with Plecter fonts modified with Hum Extension tool.

AirmakZ, 2018-03-05:
Are you using DFPlayer from eBay, AliExpress or similar? If so, check if the device mounted in the module is an YX.

DonSaber, 2018-03-05:
Actually it's a Prime 1.0, no homebrewing for me ;)

jbkuma, 2018-03-05:
With this and your other issue together it sounds like the sounds aren't
configured properly or the card wasn't formatted and populated properly.

if you look at the sound config file, the NR is the number of sounds,
OFFSET is the overall position in the copy order, so if you change the
number of sounds, the offsets need to be changed as well.  For instance if
you only use 1 "POWERON" sound instead of 4, then you need to reduce
POWEROFF_OFFSET and each subsequent OFFSET by 3.  Alternatively, you can
make duplicates of the sound to fully populate the default font load out.

    #define SF_BOOT_OFFSET 1
    #define SF_POWERON_OFFSET 2
    #define SF_POWEROFF_OFFSET 6
    #define SF_SWING_OFFSET 8
    #define SF_CLASH_OFFSET 16
    #define SF_LOCKUP_OFFSET 24
    #define SF_BLASTER_OFFSET 25
    #define SF_MENU_OFFSET 29
    #define SF_HUM_OFFSET 30

    #define SF_BOOT_NR 1
    #define SF_POWERON_NR 4
    #define SF_POWEROFF_NR 2
    #define SF_SWING_NR 8
    #define SF_CLASH_NR 8
    #define SF_LOCKUP_NR 1
    #define SF_BLASTER_NR 4
    #define SF_MENU_NR 1
    #define SF_HUM_NR 1

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

DonSaber, 2018-03-05:
Good point, but I don't believe this to be the cause of my problems.
I did not change the structure nor the number of files in the soundfonts, they are just the same numbers of fonts as the default package.
The only difference is that I currently have 6 sound fonts in the SD (properly configured in the code).
I also tried to reduce the number of fonts to 3 with not noticeable improvements.
Besides, the problem is occasional, normally I can browse through the fonts in the menu with the proper font file being played. I can select the font I want and when the saber turns on, it plays just fine until something happens, which might be 10 minutes later or 30, not clear.
I expect that if the structure of the SD or the number of files is wrong, the font should not work at all since the beginning.
These are two tricky issues, not easy to troubleshoot as I cannot spar with the debug cable plugged in :)

jbkuma, 2018-03-05:
The random functions are not perfectly random, so if they are missing a
file or only occasionally playing the wrong one the flaw in the function
can sometime mask the issue and seem like something else.  Since it's
occasional and random it seems that it would be tied to a memory issue or
infrequent math error.

When you same hum keeps playing.. does the power off sound play first?

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2018-03-05:
Buongiorno Don, I can only repeat what jbkuma already wrote, wrong sounds played are always an indication of wrong order. Did you select the board type correctly (i.e. Prime and not Stardust)? They have differenced in the sound font structure. Try to upload the sound image available as a zip file under https://github.com/Protonerd/FX-SaberOS/tree/master/soundfonts.

DonSaber, 2018-03-05:
I'd really like it is an SD problem, but I still have my doubts.
I'm attaching a text file with the content of the SD card, so you can double check what I did.
[SD_Structure.txt](https://github.com/Protonerd/FX-SaberOS/files/1782793/SD_Structure.txt)
Basically, first I created the directories in the SD, then moved to the Config directory and drag-and-drop the files to write them in order. Then I repeated the process for the all font directories.
As far as I understand I should have done it in the correct way, but please check it.

jbkuma, 2018-03-06:
It looks correct.  FYI, you don't really need overall consecutive
numbering.  I usually just drop over the folders for the fonts I want in
the order I want them, the copy order is the most important part, names are
totally irrelevant.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2018-03-11:
Is this still an open issue? Please let us know if you still need help!

DonSaber, 2018-03-13:
I updated the saber to the latest build but last week I had no chance to test it extensively.
Hopefully I should be able to put the saber under pression tomorrow night.
Will let you know.

DonSaber, 2018-03-22:
I wasn't able to test the saber really extensively in the last two weeks, but the problem did not occur again after updating to the latest build.
I'd say we can close the issue for now.
I'll open a new one if I get the issue reproduced again.

closed 2018-03-22 by DonSaber