Protonerd/FX-SaberOS #68: speaker buzz and no sound playing

luke2636, 2018-03-09:
im making a homebrew lightsaber with nano dfplayer and mpu6050 , i have wired it as per the homebrew led star schematic and im getting no sound just a load buzzing sound from the speaker

jbkuma, 2018-03-09:
You either have a power or grounding issue.  Make sure your battery is
charged and that none of your solder joints are shorting.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

luke2636, 2018-03-09:
im powering directly from my laptop via ftdi usb sorry this is my first electronic project so i am a noob , im using a breadboard at the mo and jumper so should have not sorting and the solder joints seem good on all of my parts

luke2636, 2018-03-09:
okay thanks i will give that a shot and hopefully that will work

Protonerd, 2018-03-11:
I can confirm, USB will not be able to supply the board with a premium speaker connected to it at full volume. A homebrew will make the situation even worse. Easiest option to try is either to select a better supply (like jbkuma suggested, or simply hook up the battery), or lower the volume of the DFPlayer to about 15.

luke2636, 2018-03-15:
thanks for all your advice i have connected a 5v 1a power supply, and buzzing has ceased but now i  have some more issues i have a hum playing all the time even before i ignite the blade once ignited clash swing blaster and lock up interrupts work but the hum will only last 5 second afterwards
 even tho the hum file in the sound file is 16 seconds long. thanks again for your help so far.

[image]

[image]

Protonerd, 2018-04-05:
The code restart the hum sound every 5 secs if the default settings are used. You only can have problems if your hum or hum extension is shorter than 5 secs. Otherwise regarding the problem of hum playing, it is a copy error to your SD-card.

luke2636, 2018-04-12:
The sound file is the diyino prime image I dragged and dropped the content to the sd card after a full format to fat32 I'm using Windows10 and problem persist I can't seem to find what I am doing wrong I have ordered a new dfplayer from banggood to see if the issue lies there as mine is a Chinese copy from eBay