Protonerd/FX-SaberOS #69: Compiling errors when enabling SOFT_ACCENT

DonSaber, 2018-03-28:
Yesterday I was testing the SOFT_ACCENT feature and I got two different problems when compiling the build.

The first issue was raised against fadeAccent(), that was not declared.
I solved the problem adding the following definition in the light.h file:

76 void fadeAccent();

Right after fixing the fadeAccent(), I got the following error messages, which I'm not able to understand and solve:

Arduino: 1.8.5 (Mac OS X), Board: "Arduino Nano, ATmega328"

sketch/ConfigMenu.cpp.o (symbol from plugin): In function `confParseValue(unsigned int, unsigned int, unsigned int, short)':
(.text+0x0): multiple definition of `pwmPin'
sketch/Buttons.cpp.o (symbol from plugin):(.text+0x0): first defined here
sketch/FX-SaberOS.ino.cpp.o (symbol from plugin): In function `MPU6050::dmpInitialize_light()':
(.text+0x0): multiple definition of `pwmPin'
sketch/Buttons.cpp.o (symbol from plugin):(.text+0x0): first defined here
sketch/Light.cpp.o (symbol from plugin): In function `lastAccent':
(.text+0x0): multiple definition of `pwmPin'
sketch/Buttons.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Multiple libraries were found for "Wire.h"
 Used: /Users/smadama/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/Wire
 Not used: /Users/smadama/Documents/Arduino/libraries/Wire
Multiple libraries were found for "DFPlayerSerial.h"
 Used: /Users/smadama/Documents/Arduino/libraries/DFPlayer_LSOS
 Not used: /Users/smadama/Documents/Arduino/libraries/DFPlayer
 Not used: /Users/smadama/Documents/Arduino/libraries/DFPlayer
 Not used: /Users/smadama/Documents/Arduino/libraries/DFPlayer
 Not used: /Users/smadama/Documents/Arduino/libraries/DFPlayer
exit status 1
Error compiling for board Arduino Nano.

jbkuma, 2018-04-03:
> Can anyone look into this problem ?
> It is still occurring with the latest build.

Don't use soft accent.

DonSaber, 2018-04-03:
Well, it seems quite a drastic answer.
Is there any plan to complete the feature in the near future ?

Alternatively, can someone at least give me suggestions to try and understand why I get these "multiple definition of" errors?
I may try to work on it on my own, if I know what to look for.

jbkuma, 2018-04-03:
I wanted to get you an answer after a couple days waiting and didn't have
time for a longer response.  This feature has never worked since the
original LSOS.  I have removed it in the past but it seems to have snuck
back in.  I am working on adding pixel accent functions that will allow an
arbitrary number of 5mm or other WS2812 compliant LEDs.  I will have a demo
of my current progress on our Facebook group late this evening (NYC time).

One of our users is currently working on correcting the minor errors that
deal with defining defaults in functions.  If these are the errors you are
referring to, they can be ignored and should not prevent the sketch from
compiling.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Protonerd, 2018-04-05:
I can secound jbkuma, no one ever used this function, I also see no value in it. I also removed it already, no idea how it sneaked back again... Anyway, with the Prime if not driving segmented LED strings you have ample PWM channels available to make your soft accent with them, so no need for these functions.

DonSaber, 2018-04-10:
Sorry for replying so late, but last week I had a little surgery and only now I find some time for other matters.
The idea of driving accents with Pixels is nice, but I guess it won't be compatible with the sabers using Star Led setup, unless we can enable pixel support also in this case.
Anyway, I will not use anymore the Soft-Accent, however I still like the idea of having a soft-pulsing accent LED.
I will see if I can find some kind of solution, and in case will let you know.
I leave the post open in case anyone may want to add something, elsewhere it can be closed.

jbkuma, 2018-04-10:
It will be 100% compatible with Star LED setups.  Even an illuminated AV
switch can work as a pixel either by modifying the switch with a 5mm pixel
LED or adding a WS2811 chip which can drive it as a single color at 18ma.

You can see a preview of how it will work here:
https://youtu.be/rnMnIR5JP5E?t=39s (mainly just the top pixel, the others
are mostly proprietary decorative animations , except for the meter modes )

The pixel accent can also be used as a pilot for gravity color mixing, so
it will make that more viable for Star LED setups.

Of course we can't account for all possible accent configurations
(effectively infinite possibilities) so the basic modes will be mirror and
meter. In mirror all LEDs will function the same way, in meter certain
functions will have bar graphs and animations if there is more than one
accent. The implementation will at least provide a starting point from
which the programmer can add custom functions.

If you are on Facebook I will be announcing when it's ready there, and I've
already made some posts about it.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe