Protonerd/FX-SaberOS #71: Power supply/inside charging

spazone, 2018-03-31:
Hello everybody!
I’m building a homebrew-luxeon system.
If powered by a single 18650 battery everything works fine, but LED power is too low (almost non visible). So I’m trying to use two 18650 connected in series. I’m using a WH-A2S80A battery protection circuit to charge the batteries in the hilt (without removing them). With this configuration I can’t make the system to work.
If powered by the two batteries alone everything works fine, but when I connect them through the protector  I get a 4.(something) v reading (output reading straight from the protector circuit: 7,6v ; output reading when connected to the system 4,5v)
I can’t find what’s wrong, with this config led works way better, and I can charge batteries inside the hilt without removing them while also getting a killswitch.
MOSFETS are irfz44n, im using VIN pin as input, DFPlayer mini, and mpu6050.
 I’m stuck and losing it lol, so any help would be awesome!
Let me know if you need more details or pictures.
Thanks in advance (sorry for my English as it’s not my mother tongue)
Bye!

Himmiko, 2018-04-03:
Привет! пришли фото схемы зарядки. У меня есть предположения.

spazone, 2018-04-03:
Hi!
I don’t know how to embed images, but here’s a link:
http://www.icstation.com/images/uploads/9845_2.jpg

spazone, 2018-04-03:
I’m thinking about using protected batteries, run them straight to the system, a “parallel” connection to the charging circuit, and a jack used as a killswitch to switch between them.
You guys think this is ok?

jbkuma, 2018-04-03:
One 18650 should be sufficient. If it is not something is wrong. The LEDs
sound only need around 2-3.5v depending on the color. Resistors should be
required to lower the voltage, not more batteries.

How have you wired this?

spazone, 2018-04-03:
I tested it on a breadboard and light output was extremely low (you could actually look straight at the led without going blind), I’m using an Arduino Nano clone, dfplayer mini and mpu6050.
[image]

jbkuma, 2018-04-03:
The channels work essentially just like a Nano Biscotte or NEC board, the
LEDs should be receiving nearly the full battery voltage through the
MOSFETs.  There is going to be some loss through your breadboard and
jumpers.  Your photo does not show the LED.  It looks like you are only
using two channels with only the ground coming from the breadboard, is the
other side connected directly to the battery? What size resistors are you
using? Is it wired in series or parallel?  Have you tried changing the
color profile?

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Himmiko, 2018-04-03:
Моё предположение в том,что при подключении через защищенную плату вам просто не хватает силы тока. у меня была похожая проблема. я просто отказался от использования платы внутри рукояти. а вместо этого сделал внешний блок. батареи остаются внутри,но заряжаются через ключ

spazone, 2018-04-03:
I was testing the different battery configurations.
With one battery mpu6050 is in series by 3v3 pin, the rest is in parallel; with 2 batteries in series mpu6050 and dfplayer come from the nano and the led are in parallel.
One of the leds is disconnected because I was measuring voltage supplied (a little bit over 2v with a single battery).
Ground and + are both coming from the 2s battery charging circuit or straight from the single battery . I took that pic as a guide to remember some wirings.
I tried several color profiles, and changed the brightness config from the software.
I wasn’t using resistors after the led, as I was just testing the wirings. Resistors on the mosfets are 220 ohm (+- 5%).

spazone, 2018-04-03:
Batteries are 18650 2600mA

jbkuma, 2018-04-03:
Your LEDs need to have resistors following ohms law.  V=IR, R= I/V where V
is the battery voltage minus the forward voltage of the LED and I is the
desired current.

If you are only getting 2v and the profile is using a fully bright channel
for the LED, the problem isn't the battery, it's something else.  You could
be loosing too much power through the breadboard and your wiring, or your
battery may need to be charged.  Check the voltage of the battery with
everything turned on.  The resistors on the mosfet are only necessary to
pull down the gate, they don't affect output voltage.

If you aren't using a matched pair of batteries (exact same charge level
and capacity) in your 2s configuration you can end up with big problems.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

spazone, 2018-04-03:
Battery is 3.76v , I’m going to try with the soldered system and only one battery to check for improvements.
I’m using paired batteries with a charge/discharge protection circuit:
[image]
I’m running out of ideas LoL. Do you think the mosfets I’m using are ok?

closed 2019-10-23 by jbkuma