Protonerd/FX-SaberOS #72: Blaster Deflect

FatherCool, 2018-04-03:
Trying out the new sketch and its great, however now I cannot use the double click on the aux button to activate motion blaster deflect. I loved the feature before. Any ideas?

jbkuma, 2018-04-03:
In the most recent update double click has been disabled and the effect is
triggered by a single click on the activation button.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

FatherCool, 2018-04-03:
Thanks for that. Works like a charm. I actually prefer it this way as its easier to activate and I never used the lock up on clash before anyway, preferring to do the lock ups manually with the aux button.

jbkuma, 2018-04-03:
We decided that blaster block was more fun and gave it priority.  Glad you
enjoy it!

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2018-04-05:
Activate Lockup ?? in this versión

Protonerd, 2018-04-05:
single or 2-button mode? In single button mode lockup has been removed at the same time I retired the double click. If using 2-buttons, you can still activate the lockup with the aux button. Clash triggered lockup should also be possible, AFAIK.

shiryu55, 2018-04-05:
that bad . I liked the option of blocking, there would be some way to have that function with a single button, such as pressing the button for 2 seconds, the saber that I am building has only one button.

shiryu55, 2018-04-05:
http://www.youtube.com/watch?v=KnBN2dRgvsE

jbkuma, 2018-04-05:
The lockup on clash is disabled in 2 button mode, I have some code I'm
working on where I re-enabled it.  When I finish the pixel accent code I'll
include the fix.  I think we can add in the double click for single button
too, and just use it for this feature.  The clash menu navigation is very
superior to the double click method, but I think leaving it for mode
activation should be fine: single click for blaster blocks, double click
for lockup on clash, long press to power off the blade.

There is also the possibility of using long press to activate modes in the
standby state.  This might be a handy place to put in a mute function.  In
my current build I'm using long press in standby for a quick font change.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

shiryu55, 2018-04-05:
Hi mate but i dont understand wich is the function to activate the new lockup clash and blaster deflet in this version

closed 2019-10-23 by jbkuma