Protonerd/FX-SaberOS #74: Clash detection doeesn't work

AndongYi, 2018-05-11:
For some reason any clash detection dissaperared (swing detection is still there) in last version. if I single click a button saber goes into blaster block mode and double click does nothing. 

Tried reloading previous version of FXSOS with disabled lockup clash in singlebutton setup and clash detection works fine so i assume both wiring and MPU are fine.

Using arduino nano, df player, MPU6050, 3 strips of neopixel and a single 18650 for power

jbkuma, 2018-05-11:
There are two modes for clash detection. If you are configured for
interrupt based clashes then the int pin of the mpu must be wired to D2,
otherwise use the polling method.

AndongYi, 2018-05-11:
SInce i'm a total noob in coding is there a simple way to change this menthod in FXSOS or it would be easier to just wire int pit to D2?

Protonerd, 2018-06-29:
INT based clash detection is supposed to react instantenously, but in reality you will hardly see any difference, both work spot-on.

closed 2019-10-23 by jbkuma