Protonerd/FX-SaberOS #75: Soundfonts error

peterdc201, 2018-06-02:
i have gotten the soundfonts onto the sd card i have and yet it dosen't work with homebrew lightsaber i hear a blaster sound for boot up and it dosen't use the correct hum sounds instead i hear a r2d2 over and over and over anyone know why this happends

jbkuma, 2018-06-02:
The card isn't formatted or loaded correctly.

http://opensaber.proboards.com/thread/8/prepare-sd-card

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

peterdc201, 2018-06-02:
i tried to follow your instructions i reformatted the card i extracted the files without copying them and it still dose the same

peterdc201, 2018-06-10:
i tried again and it is still doing the same i reformated moved the files without copying yet it is still giving me a bad time

Beskor, 2018-06-28:
Everything was done as it should, but the last sound is still repeated all the time...

Protonerd, 2018-06-29:
Which sound exactly?

peterdc201, 2018-06-29:
for me all of them

peterdc201, 2018-06-29:
it keeps mixing up the sound fonts and it won't go into boot mode and a lot more

peterdc201, 2018-06-29:
sry about that i acidentaly hit thew wrong button

Beskor, 2018-07-01:
The first sounds boot sound as it should be, then I turn on the saber, all other sounds work as it should, except for clash sound (dont know why), and when I turn off the saber, the last sound begins to repeat itself infinitely and all subsequent sounds in the config mode/

AirmakZ, 2018-07-01:
The ic device mounted in your Dplayer module is not the original one. It’s the cause of this wrong behaviour

Beskor, 2018-07-01:
Any ways to fix it? Or buy the original df player?

AirmakZ, 2018-07-01:
In theory your module uses a red led. that’s right?
The good ones are mounting blue LEDs and also there are differences in the pcb silkscreen.
This is good one. You can see white squares around SMDs
http://s.aliexpress.com/BveIZVri

Beskor, 2018-07-01:
You are right, the LED was red, and there was no white squares, I looked in the warehouse and found one such as you described, with a blue LED and squares. I moved them around, but the result was the same.

jbkuma, 2018-07-01:
> except for clash sound (dont know why),

We've added a second clash method that uses the interrupt on the MPU6050 to D2 (interupt method) the old method is polling.  You can change this in the config.

peterdc201, 2018-07-02:
i just looked through the messages and my led is also red iv ordered a blue dfplayer led with the white outline i hope it workes for me like it did for you beskor

AirmakZ, 2018-07-02:
I think this issue is going to be quite common...
@jbkuma, maybe is good idea to add a comment in the Readme.

jbkuma, 2018-07-02:
There is already a comment in the config files, we should probably change the default method to polling since it is the traditional method.

```
/*
 * CLASH DETECTION METHOD
 * Clash can be detected by programming the mpu to trigger an interupt
 * on motion. If a motion interrupt is triggered, it can either be
 * detected by connecting the MPU's INT signal to D2 of the AVR (-> CLASH_DET_MPU_INT)
 * or by polling the INT_STATUS register via I2C and determining the value (-> CLASH_DET_MPU_POLL)
 * of bit 6
 */
#define CLASH_DET_MPU_INT
//#define CLASH_DET_MPU_POLL
```

Protonerd, 2018-07-03:
I thought all the while that the issue with the repeating retraction sound is a fault of fake DFPlayers, until recently I had this issue myself too. It happened exactly 3 times over a period of almost 3 years and a lot of boards I tested, but funnily if I look into the code I must conclude that it should happen in fact all the time (and probably we would have eliminated this bug a long time ago). I will look into this and solve it, should not be very complicated.

peterdc201, 2018-07-03:
i hope so cause i really would love it to work and start mounting leds on it and work on the chassis

peterdc201, 2018-07-14:
k i just installed the new df player its still mixing up the sounds

Himmiko, 2018-07-23:
Hi guys. I have the same problem. When the saber is switched on, the sound p2d2 is played. Previously, everything worked perfectly .. can anyone have an old version of the firmware? or somehow you can fix this?

JagdNutCrkr, 2018-08-11:
For what it's worth, I've been able to verify what AirmakZ and Protonerd suggested about the MP3 module.  I've tried 2 different sellers, both times the module had the blue LED and white squares around the components, but I still ran across looping issue.  On a previous post, Protonerd mentioned that the MP3 module that has the YX5200 MP3 chip is the original DFPlayer.  On the website, [,](http://www.flyrontech.com/index.html) they claim to be the original manufacturers of the YX5200 chip.  I also saw the YX6300 chip which is what is in the Stardust V3 board.  So I decided to give them a shot, bought a DFPlayer from one of their partnered distributor [.](http://www.flyrontech.com/eabout/3.html)  The DFPlayer or FN-M16P as marked from the US distributor, also has the blue LED and the white squares around the components.  TLDR: the MP3 module from that seller/manufacturer resolved the looping issue for me.  Hope this helps everyone =)

JagdNutCrkr, 2018-08-17:
You're absolutely right, AirmakZ, the sellers do not advertise which IC they are using. I actually bought one that showed the correct IC on the store page but received a module without the right IC.  I bought two more modules from the seller I mentioned, I'll let you know if those are good too.
I also thought about replacing the IC on the bad modules since flyrontech do sell them, but my soldering skills aren't there yet =)

JagdNutCrkr, 2018-08-19:
Happy to report the additional MP3 modules I ordered are also good =).  Seller is a keeper!

Protonerd, 2018-09-27:
just as a side note: I pushed a new version yesterday, it should fix the repetitive rettaction sounds even for fake DFPlayers (if not, those are not fit for this application).

closed 2018-09-27 by Protonerd