Protonerd/FX-SaberOS #76: Color of the Pixel Accent not matching with Pixel Blade (v1.1-beta)

AirmakZ, 2018-08-17:
I installed the v1.1-beta to check the pixel accent amazing feature. Taking into account it's just a beta version it working really fine, but I have an strange phenomena....:

The color of the pixel blade is not matching with pixel accent. Basically I have the following relations:
Blade: red -> Accent: green
Blade: blue -> Accent: blue
Blade: green -> Accent: OFF

I don't think it's related with hardware because pixel accent if changing the color properly, even with different brightness level...

Thanks in advanced for your support!

jbkuma, 2018-08-17:
Depending on the LED red and green might be swapped, but no color should
just not work.

In FXSaberOS.ino, after line 121
  WS2812 accentPixels(NUM_ACCENT_PIXELS);
add
    accentPixels.setColorOrderRGB();

I think this should work, but I can't test it.  If it works I will update a
new version that includes a flag "swapAccentRG" that will make it easier to
fix this in the configuration.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

AirmakZ, 2018-08-17:
Hi jbkuma, thanks for your quick answer.

Your proposed solution is WORKING FINE !!
Just instead of line 122 I added the command inside setup, otherwise I have the compilation error: 
_FX-SaberOS_beta:121: error: 'accentPixels' does not name a type_

```
void setup() {
  #ifdef PIXEL_ACCENT
    accentPixels.setColorOrderRGB();
    accentPixels.setOutput(PIXEL_ACCENT_DATA);
  #endif
```

I don't know how the WS2812 library works but taking into account I'm using leds from the same reel... it seems strange.... Anyway.... SOLVED.

Regarding red color issue I will try replacing the led because I unsolder it from the strip and maybe It's damaged.

Thanks again.

jbkuma, 2018-08-18:
I added a fix to the main branch with notes on Facebook and Open Saber forum

AirmakZ, 2018-08-18:
@jbkuma, one question.
Modification has been added in master branch, but seems it's not including all pixel accent features same as beta, right?
Is there a merge activity scheduled in the future?

By the way, just confirm that red color issue has been solved replacing the led 👍

jbkuma, 2018-08-18:
Until the past couple small fixes I think the master branch matched the
beta.  Master should be considered the most up to date at the moment.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

AirmakZ, 2018-08-18:
Understood !! 👍

closed 2018-08-18 by AirmakZ