Protonerd/FX-SaberOS #78: RGB Momentary Switch and Battery Charge

SliceBraver, 2018-10-21:
Hello!
I come from Italy, i do lightsaber combat and i'm building a lightsaber with an homebrew configuration.
I need to ask if it's possible configure a RGB momentary switch (main button) to show the charge level:
100-71%: Green
80-41%: Blue
40-21%: Yellow
20%-1%: Red

Also i want to configure my LS so I know the charge level while i'm fighting or charging the battery (both in idle and action mode).
Ex: at 50% i will hear "Battery at 50%", or when the LS is full charged: "Battery full charged".

Tell me if you are working on those features or maybe i can try to add some strings (but i need some indications).
Thank you for disponibility and sorry for the bother (and bad english). :)

jbkuma, 2018-10-21:
The battery meter (also volume meter, font indication, gravity color pilot,
etc) are all available via the pixel accent code. If you are you using a
pixel or a pixel adapter board with your switch you can easily use this.
The meter is on a green-red-yellow range.  If you want to add blue, I
suggest you add it as your "full" color, as it is better logically.

As for your specific function, we don't have any plans to add such
features, but it should be possible to do these things without too much
trouble using only existing functions.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

SliceBraver, 2018-10-22:
Unfortunately, i don't use a pixel adapter.
Using a star led i have some free channels on my Arduino nano so i was thinking of use it to wire my RGB momentary switch and control the light colors via code.

For the voice notification the charge level:
Adding some audio file on SDCard (like 50% battery charge) don't give some troubles with files index?
I saw that audio reading works on SDCard files index.
Another question: where i have to write these string? (i need it both in idle that in action mode)

Thank you for your disponibility.

SliceBraver, 2018-10-22:
I've found the "batLevelConfigEnter()" function, with this code (i hope i don't do something wrong copying part of the code):
```
      if (batLevel > 95) {        //full
        SinglePlay_Sound(19);
      } else if (batLevel > 60) { //nominal
        SinglePlay_Sound(15);
      } else if (batLevel > 30) { //diminished
        SinglePlay_Sound(16);
      } else if (batLevel > 0) {  //low
        SinglePlay_Sound(17);
      } else {                    //critical
        SinglePlay_Sound(18);
      }
```
Now i will use (and modify) this part to develop what i need, but the question is:
Must i insert this under: 
`else if (SaberState == S_STANDBY) {`
and
`(SaberState == S_SABERON) {` ?

And now i have to invent something to manage the SDCard files problem... ^^

jbkuma, 2018-10-22:
To add or change sounds you must also adjust the soundfont.h file.

As to the code changes to want to make, you probably want to add some
checks to the main loop function.  The easiest thing to do will probably be
to disable the built-in accent code and write your own functions.

SliceBraver, 2018-10-23:
Thank you a lot for your patience!
I apologize if i'm insistent but i don't know well FX-Saber code so i ask you for a confirm.
Also, i'm awaiting the battery for my LS and i can't test the code. :( 

I tried to write code for the RGB Accent Led:

In "Config_HW.h" i changed the "HARD_ACCENT" part with this:
`  #ifdef HARD_ACCENT
        #define ACCENT_RED 9 
	#define ACCENT_GREEN 10 
	#define ACCENT_BLUE 11
  #endif`

After i added this in Light.cpp, under "accentLedControl" function:
```
void accentLEDControl( AccentLedAction_En AccentLedAction) {
#if defined HARD_ACCENT
  if (AccentLedAction==AL_PULSE) {
        int batLevel = 100 * ((batCheck() - LOW_BATTERY) / (FULL_BATTERY - LOW_BATTERY));
        if (millis() - lastAccent <= 400) {
          if (batLevel > 75){
            analogWrite(ACCENT_RED, 0);
            analogWrite(ACCENT_GREEN, millis() - lastAccent);
            analogWrite(ACCENT_BLUE, 0);
          } else if (batLevel > 50){
            analogWrite(ACCENT_RED, 0);
            analogWrite(ACCENT_GREEN, 0);
            analogWrite(ACCENT_BLUE, millis() - lastAccent);
          } else if (batLevel > 25){
            analogWrite(ACCENT_RED, millis() - lastAccent);
            analogWrite(ACCENT_GREEN, (millis() - 42) - lastAccent); // lesser green (255-42=213) to create orange.
            analogWrite(ACCENT_BLUE, 0);
          }else {
            analogWrite(ACCENT_RED, millis() - lastAccent);
            analogWrite(ACCENT_GREEN, 0);
            analogWrite(ACCENT_BLUE, 0);
          }
        } else if (millis() - lastAccent > 400
            and millis() - lastAccent <= 800) {
            if (batLevel > 75){
            analogWrite(ACCENT_RED, 0);
            analogWrite(ACCENT_GREEN, millis() - lastAccent);
            analogWrite(ACCENT_BLUE, 0);
          } else if (batLevel > 50){
            analogWrite(ACCENT_RED, 0);
            analogWrite(ACCENT_GREEN, 0);
            analogWrite(ACCENT_BLUE, millis() - lastAccent);
          } else if (batLevel > 25){
            analogWrite(ACCENT_RED, millis() - lastAccent);
            analogWrite(ACCENT_GREEN, (millis() - 42) - lastAccent); // lesser green (255-42=213) to create orange.
            analogWrite(ACCENT_BLUE, 0);
          }else {
            analogWrite(ACCENT_RED, millis() - lastAccent);
            analogWrite(ACCENT_GREEN, 0);
            analogWrite(ACCENT_BLUE, 0);
          }
        } else {
          lastAccent = millis();
        }
  }
  else if (AccentLedAction==AL_ON) {
          int batLevel = 100 * ((batCheck() - LOW_BATTERY) / (FULL_BATTERY - LOW_BATTERY));
          if(batLevel > 75){
          analogWrite(ACCENT_RED, 0);
          analogWrite(ACCENT_GREEN, 255);
          analogWrite(ACCENT_BLUE, 0);
        } else if (batLevel > 50) {
          analogWrite (ACCENT_RED, 0);
          analogWrite (ACCENT_GREEN, 0);
          analogWrite (ACCENT_BLUE, 255);
        } else if (batLevel > 25){
          analogWrite (ACCENT_RED, 255);
          analogWrite (ACCENT_GREEN, 213);
          analogWrite (ACCENT_BLUE, 0);
        } else {
          analogWrite (ACCENT_RED, 255);
          analogWrite (ACCENT_GREEN, 0);
          analogWrite (ACCENT_BLUE, 0);
        }
    //digitalWrite(ACCENT_LED,HIGH);
  }
  else {  // AL_OFF
    analogWrite(ACCENT_RED, 0);
    analogWrite(ACCENT_GREEN, 0);
    analogWrite(ACCENT_BLUE, 0);    
  }
#endif //HARD_ACCENT
}
```

Logically i think it can works, but unfortunately i don't know very well FX-SaberOS and i can't test it.
What do you think about it?

closed 2019-10-23 by jbkuma