Protonerd/FX-SaberOS #80: Power On Sound duration

Danvader389, 2018-12-07:
Hello,

I am building an homebrew version of FX-SaberOS. 
I am using the defaullt DIYino_Prime_SDCard_soundfonts
When I press the button to activate the lightsaber It is like the power-on sound is shorter that needed or too fast. However the power-off sound duration is perfect.
Is there a way to fix this problem ?

Thanks for helping

jbkuma, 2018-12-07:
Edit the duration in soundfont.h

Danvader389, 2018-12-08:
Thanks jbkuma for your answer
I have another question: Is the resistor between gate and source of mosfet optionnal or mandatory?
I am using FQP30N06L mosfet

Protonerd, 2018-12-10:
No, they aren't necessary, the code takes care about controlling the gates in sleep/idle modes, so they are never undefined.  The Stardusts do not have them either any more.

StrangeD0s, 2019-03-21:
I also have a question about the power on sound that might fit in here. Will changing the duration in soundfont.h also affect the lighting up speed of the neopixel? I am currently trying to make a power on sound that is fast an snappy but has some additional sizzling going on right after the blade has extended. But I never seem to be able to align the length of the sound with the speed of the neopixel ignition. If I make the duration in soundfont.h shorter, it seems to cut the power on sound off and go straight to hum.

jbkuma, 2019-03-21:
The duration controls the speed of the animation, and the hum does begin
immediately at the end of the animation.  Originally it didn't cut the
sound off, but I'm not sure what changed.

There are 2 factors that will affect this: HUM_RELAUNCH and PowerOnTime.
HUM_RELAUNCH defaults to 2000ms, and PowerOnTime is the timing of the
animation.  What is supposed to happen is that the hum relaunch is
suppressed for PowerOnTime+1000 which should be plenty of time.

This is the vital part to relaunch the hum in the main .ino
[image: image.png]

I don't have something I can test this with at the moment, but you could
try changing the if statement to
 if (((millis() - sndSuppress > HUM_RELAUNCH) && !(hum_playing))) {

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

StrangeD0s, 2019-03-22:
Thanks for your answer. Your linked image is not showing up, but I think I understand it better now.

closed 2019-10-23 by jbkuma