Protonerd/FX-SaberOS #81: Compilation error

spazone, 2018-12-31:
Hello everyone!

I have absolutely no coding experience (I'm actually a chemist), and after following all the steps to set up the arduino IDE in the wiki (I believe properly, but evidently not) I'm receiving this error when compiling:

[image]

I have absolutely no idea what could be wrong, so any help would be greatly appreciated!!

spazone, 2018-12-31:
You mean instead of copying all of them to arduino libraries folder? (I copied all of them to arduino’s folder, and kept them as unzipped in the master folder of FX saberOS)
Thanks for the quick reply!!!

Happy new year!!!

per1234, 2019-01-01:
Which version of the Arduino IDE are you using?

spazone, 2019-01-01:
Thank you all for your support!
I'm not really sure what was wrong, but I uninstalled IDE (seems like I had an old folder somewhere and it kept searching in both folders), started everything from blank, left libraries in project folder and in arduino libraries folder and.. it compiled!!!!
I'm about to try it now, happy new year everybody!
(I installed Arduino IDE app from Microsoft store)

per1234, 2019-01-01:
You must have been using a version of the Arduino IDE older than 1.6.10. FX-SaberOS is not compatible with those IDE versions because they require that sketches contain `#include` directives for library dependencies, such as Wire.h. That could be fixed by adding this line to the sketch:
```c++
#include <Wire.h>
```
but I don't know whether it's worth the effort and clutter to support outdated IDE versions. It might be worth adding a note to https://github.com/Protonerd/FX-SaberOS/wiki/Setting-up-Arduino that specifies that the latest IDE version must be used.