Protonerd/FX-SaberOS #85: Hum sound in last foundfont slot not playing

StrangeD0s, 2019-02-18:
I have a weird issue. I use the Arduino Nano and everything works fine. Except that whatever soundfont I put in the last slot will not play the hum sound.
I initially used the DIYino_Prime_SDCard_image and there the hum sound from the FallenSon soundfont, which is in slot 5, would not play. Than I experimented and switched the FallenSon and InVader soundfonts so InVader would be in slot 5. And then the hum of InVader would not play while FallenSon now works fine. Considering that the hum sound is always the last sound in the soundfont folder and the issue only occurs in the soundfont that is in the last slot I must conclude that for some reason it is the last file on the SD card, that is not found. Has someone else experienced this issue?

ingcsmoreno, 2019-04-20:
I think I'm experiencing the same issue as you are. Only difference is that in my case the only soundfont I've copied into the SD Card is VanillaLyte, and so I don't have HUM sound at all since my first soundfont is algo my last one. 

I'll copy the rest of the sound founts and see what happen. Thanks @StrangeD0s !

jbkuma, 2019-04-20:
Have you configured as using only one font, or did you put multiple copies
of the same font on the card? What board are you using, and which
configuration did you set it to? (Prime, Stardust, etc)

Make sure you've copied the appropriate font to your SD card and reformat
when you make changes.

ingcsmoreno, 2019-04-20:
> Have you configured as using only one font, or did you put multiple copies of the same font on the card? What board are you using, and which configuration did you set it to? (Prime, Stardust, etc) [...]

hmm Interesting, I didn't know there was a setting to specify the number of sound founts, I just copied the one I would use. My SD card only has the configmenu sound files (in root), and the **SF01_VanillaLyte** folder with its files. Should I change something in Config_SW.h or reformat the SD card and copy all of the soundfont folders this time?

Regarding my board, I'm using an **Arduino Nano + MPU6050 + DFPlayer Mini** (Not an XY original, but it has a blue led, and all the white marks around the soldering so I think it should be OK), and I've set it to **DIYino PRIME**.

ingcsmoreno, 2019-04-21:
So, I've reformatted the SD card, and copied everything again, but this time I copied all the sound founts folders included in Git (not just the VanillaLyte like the first time), and the HUM **is working** for VanillaLyte, but it doesn't work with FallenSon (which is the one in the last position).

StrangeD0s, 2019-04-21:
You are experiencing exactly what I do! It is always the soundfont in the last position where the hum is not working. I also have the same hardware configuration. It would be great if we could figure out, what the problem is.

closed 2019-10-23 by jbkuma