Protonerd/FX-SaberOS #86: Popping noise in self made hum loop

StrangeD0s, 2019-03-05:
Ok, this might not be an issue with FX-SaberOS but I'm hoping to find help here anyway. I'm currently making my own soundfont and there is a popping sound in the hum that occurs every time the loop restarts. I have tried every method of creating a good sound loop. I'm using Audacity and the sound starts and ends at the zero crossing. It is exactly 7 seconds long. In Audacity there is no audibly popping sound whatsoever. But as soon as I try it out on the Arduino board it pops. I have even tried splitting it at 3.5 seconds and copying it in reverse so that it will definitely match up perfectly. But still no luck. Does somebody have any other tips?

jbkuma, 2019-03-05:
Make sure you are not saving with meta data.

StrangeD0s, 2019-03-05:
I just tried something completely different: I added another second at the end and now the pop is gone!

StrangeD0s, 2019-03-05:
Thanks jbkuma, that must have been the problem all along!

closed 2019-03-05 by StrangeD0s