Protonerd/FX-SaberOS #90: Brand New to Arduino

fadroid, 2019-10-24:
is it possible to to just wire up the Arduino nano without the dfplayer and mpu6050 if you don't want to utilize sound and motion? My goal is to start with a neopixel blade and I just really care about the ability to change colors. Thank you in advance!

jbkuma, 2019-10-24:
I recommend following the adafruit guide if that is what you want.

fadroid, 2019-10-25:
Thanks for the quick response, was asking because I am trying to eliminate issues one by one and would have been happy with just light and no sound. Ultimately I would like to use all the capablities of FX-SaberOS but I can't seem to get it working. It is the homebrew set up and everything is wired per the diagram and the code uploaded correctly to the Nano.  I believe it is an issue with the main button. If you are using a button without an LED is it just wired from ground to D12 or do I also need to connect 5v and a resistor? I am using a single 18650 battery and when I attempt basic examples from the FastLED Library the 2 WS2812b strips light up without issue.

closed 2019-12-22 by jbkuma