Protonerd/FX-SaberOS #91: SD card reader.

Chattan-jedi, 2019-12-22:
This is less of an issue than a question. (Sorry new to github and couldn’t find a discussion section. 

First I love your setup and code it’s fantastic. 

My question is would it be able to replace the df player with an sd card reader and a Pam amp (I have these in stock and will have to wait at least 6weeks to get a df player) 

I figure I’d have to adjust the code df card reader runs on different pins etc.

My real question is being as familiar as you are with your code, can it be done. And do you have any advice in going about it. 

My other question is why do the speaker outputs from the DF run back to the nano and not just go to the speaker?

jbkuma, 2019-12-22:
The DF player is really a compact MP3/WAV player with built in amp.  The code just sends it triggers to play certain tracks, loop, stop, etc.  The main motivations for selecting the DF player at the time are that is inexpensive, has gapless playback, and a nice 3W amp.  The ATmega328 just isn't beefy enough to process sound and keep up with everything else.  Even the SAMD21 chip cannot really handle pixels and sound processing at the same time.

closed 2019-12-22 by jbkuma

Chattan-jedi, 2019-12-22:
Thanks. I agree the Df player is a great piece of kit. 

Dr Shane Mackintosh
DOCTOR OF CHIROPRACTIC
B.Sc (Chiropractic), B.Chiropractic

NET Practitioner 
NOT Proficiency Rated Practitioner 
NOT certified Instructor 

18 Holdsworth Street, Fremantle WA 6160
Ph: 0411066109
E: shane@drshane.net.au
W: drshane.net.au