Protonerd/FX-SaberOS #92: Button

fadroid, 2020-01-07:
Hello,

A question on button wiring for a homebrew set up with a pixel blade. If I just want a single momentary button without an accent led, is it just ground on one end and d12 on the other? Do I still need to attach anything to A0?

Thanks in advance, this project is some awesome engineering!

jbkuma, 2020-01-07:
The only thing that gets connected to A0 is an accent LED. It does not need
to be in the switch.  If you are not using an accent LED you do not need to
wire A0 to anything.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

fadroid, 2020-01-07:
So to confirm, on d12 is simply a momentary ground signal?

jbkuma, 2020-01-28:
correct.

closed 2020-01-28 by jbkuma

fadroid, 2020-01-31:
Thanks for assistance and this project! I got everything up and running and it's awesome!