Protonerd/FX-SaberOS #93: Boot sound not playing

normanbos, 2020-01-11:
I'm in the middle of building my second lightsaber and after setting up my DIYino and re-uploading everything now instead of a boot sound I only hear a pop noise from the speaker. I have tried every combination to narrow down the issue. I have tried different SD cards (always formating and copying as described in the wiki), different arduinos, even different speakers but it is always the same. No matter what soundfont in which slot I try. No boot sound. Everything else works fine. Every other sound plays as it should.

Protonerd, 2020-01-12:
Are you using a DIYino (and if yes which type) or a homebrew? The issue is probably caused by the DFPlayer not yet oroperly initialized or even supplied at the time it receives the command to play the init. Do you see it at power up or wake from sleep?

normanbos, 2020-01-12:
I'm using a homebrew board that has worked fine the past couple of month. I did nothing to the hardware configuration. I only re-uploaded the sketch after I fiddled with some settings but wanted to reset it to its original settings. Also I have just tried it with a fully charged battery as I wanted to rule it out as a source of the problem but I'm getting the same results. The problem occurs both at power up and at wake from sleep. Could the problem be related to the DFPlayer library? I am not entirely sure that I used the one provided with the FX-SaberOS since I have used the Arduino software for a few other projects since.

normanbos, 2020-01-12:
This is the output of the serial monitor when I boot the Arduino:

`size of StoreStruct: 72
32
L03⸮⸮
EEPROM LOADED
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful
Initializing DMP...
address: 200
output: -1822

address: 202
output: 2892

address: 204
output: 1390

address: 206
output: 119

address: 208
output: -65

address: 210
output: 33

Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 0)...
DMP ready! Waiting for first interrupt...
checkpoint 0
Enabling interrupt detection (Arduino external interrupt 0)...
Connecting to DFPlayer...
Connected to DFPlayer !
`

jbkuma, 2020-01-12:
Are you holding the button down during power up? That puts it in mute mode.

normanbos, 2020-01-12:
> Are you holding the button down during power up? That puts it in mute mode.
> […](#)

No, no button is pressed during power up. And the system is not in mute mode, since all other sounds except boot are playing fine. I have now tried a clean install of the Arduino IDE and only using the libraries provided with the FX-SaberOS but that didn't help.

Protonerd, 2020-01-13:
I can confirm that this might be a genuine thing. I also had many of my props and sabers do exactly that. It seems heavily dependent on the individual chips, so it seems the DFPlayer has a quite wide spread wrt how fast it wakes up and ready for operation. Of course a generous wait time (in the range of a secound) can always be included before the intro sound is triggered somewhere at the end of the setup(). I also opserved that there is a difference in behaviour between a power cycle and waking up from sleep mode.

Protonerd, 2020-01-13:
Look here:
  /***** DF PLAYER INITIALISATION  *****/
  InitDFPlayer();
  delay(200);
 // according to debug on 3.11.2017, these 2 lines below cause the sporadic disable of sound. For audio tracker they are not strictly needed.
  //pinMode(SPK1, INPUT);
  //pinMode(SPK2, INPUT);
  SinglePlay_Sound(soundFont.getBoot());//11);

Try to increase the delay() after   InitDFPlayer(); to 1000 (which translates to 1000ms or 1sec)

normanbos, 2020-01-13:
That did it! Thank you for for looking into in and clarifying the issue. A delay of 800ms was needed for the DFPlayer to play the boot sound.
I find it still strange that I had to do that because the exact same hardware configuration worked fine with a delay of 200ms before. But now I know how to fix the issue for future projects :)

closed 2020-01-15 by normanbos