Protonerd/FX-SaberOS #94: Possible to change mainColor to being set by RGB sensor module?

normanbos, 2020-01-19:
I have an idea that I would like to try out, but I don't understand the code well enough: Could the code by changed so that the `mainColor` is defined by a RGB sensor module like the TCS34725? It could be used to read the color of a (faux) kyber crystal so that the color of the pixel blade would depend on the color of the crystal.
I have tried using the TCS34725 with some test code to read some small pieces of colored transparent plastic and making a Neopixel light up the same color. It worked really well and would be a very cool feature to further personalize a lightsaber.
But I'm not yet good enough to just add some extra code to the FX-SaberOS to achieve that.
I know the color profiles are defined in the Light.cpp and I would probably need to add some code to the FX-SaberOS.ino, probably somewhere at the start to read the RGB values and store them in `currentColor`. But that's as far as I understand it. Any pointers?

jbkuma, 2020-01-21:
I think one of the most convincing ways to do it would be to make a pressure or tactile switch that the crystal can hold closed (clicked).  If the switch is not triggered (open/not clicked) the saber is forced to stay in a standby or maintenance mode, if the switch goes to triggered (closed/clicked) the routine that checks the color activates and sets the color for the blade.  

You might be able to accomplish this pretty simply by saying something like:
[image]
(I had to paste a picture of the code for annoying formatting reasons on github)

jbkuma, 2020-01-21:
your crystal triggers probably also want to change the state engine so that an active blade gets extinguished and the saber goes into standby, etc.  You could add extra sound effects to go along with this as well, and edit SoundFont.h accordingly.

normanbos, 2020-01-21:
That is an excellent idea! Thank you! I will try to implement that. It will probably take a while since I'm just starting to learn programming and still need to find my way around functions and classes and so on. I'm not even sure where exactly to start but I'll work that out :D. I will give updates on my progress.

jbkuma, 2020-08-13:
closing this issue.  It is possible, but not currently under consideration to add to main branch.  may revisit later.

closed 2020-08-13 by jbkuma