Protonerd/FX-SaberOS #96: Crossguard Saber

fadroid, 2020-02-07:
Hello,

I have a Crossguardsaber wired up and everything except the side blades is functioning properly. I just want to make sure I understand the wiring. I currently have the side blades neopixel data lines connected together with a 330ohm resistor going to A0 on the nano, is this correct? I also have the main blade data line connected to D13 with a 330ohm resistor and that one is functional. I might be misunderstanding the pixel accent data.

I also made sure to define it as a crossguardsaber, and put the main length at 120 and crossguard at 12

Sorry to ask, but I couldn't find an answer in the forums, issues or wiki.

Protonerd, 2020-02-07:
So far you did all right, except for one thing: the sideblades form one stripe with the main blade. Therefore the side blades are connected first to D13, and the output of one side blade to the input data line of the main blade. A0 us not needed

fadroid, 2020-02-08:
Thanks for the quick response. I will update my wiring.

closed 2020-02-12 by fadroid