Protonerd/FX-SaberOS #97: Lockup on clash

fadroid, 2020-02-26:
Hello,

I wired up a 2 button homebrew setup and everything is functioning as it should except the lockup on clash. When I single press the main button it does blaster deflect on clash. The lockup effect only functions when I long press the aux button but not on clash. Has this feature changed?

Thanks in advance!

jbkuma, 2020-02-26:
Blaster blocks are on swings and clashes are on blocks. Both can be active
at the same time, and the activation is configurable.  It may be that the
most recent posted build has a bit of a different configuration than past
versions, I'm not exactly sure. If you look at the click menus you should
be able to tell how to activate each and change it if you want.

fadroid, 2020-02-26:
Thanks for the quick response! I'll take a look at the click menus.

closed 2020-02-26 by fadroid