Protonerd/FX-SaberOS #98: No audio, button doesn't work, LED strip lights up red

Rememberthecreed, 2020-05-02:
I have a video on the FX-SaberOS facebook page showing my setup. After following SirRawThunderMan's tutorial on both editing the code and loading sound fonts on a micro SD, I loaded the card into the board, and powered it on. No boot sound, the tactile switch I'm using for the main switch isn't working, and the Neopixel Strip lights up all red. I pulled the recharge port off and plugged the battery into the board, and same issue. On video I got a "Diyino Jukebox" sound to come out of the speaker, but no lightsaber boot sound. 
Short of the base code, I only edited the Config_HW.h and Config_SW.h files (as it was in SirRawThunderMan's video). Maybe the goof up is there. All other config and .h files, I left as stock. 

Config_HW.h
`/*
 * Config_HW.h
 *
 * Created on: 30 March 2017
 * author:    Andras Kun (kun.andras@yahoo.de)
 * Source :  https://github.com/Protonerd/FX-SaberOS
 * This tab contains all settings necessary to define your saber board's electronic hardware configuration
 */

#if not defined CONFIG_HW_H_
#define CONFIG_HW_H_

/* Board definitions
 *  Define what kind of Arduino compatible board you have in your saber.
 *  DIYino Prime and Stardust are both dedicated, fully integrated
 *  saber boards manufactured by ProtoWerkstatt.
 *  For more information on DIYino products, see:
 *  https://github.com/Protonerd/DIYino
 *  https://www.facebook.com/ProtoWerkstatt-740015792832712/
 *  If you have a home-brew solution using the Arduino Nano, choose DIYINO_PRIME below
 */

#define DIYINO_PRIME // including home-brew
//#define DIYINO_STARDUST_V2
//#define DIYINO_STARDUST_V3

/***** BOARD PINOUT DEFINITIONS ******/

/*
 * Definition of the MAIN and AUX buttons of your saber.
 * These 2 buttons are used to interact with the saber electronics in the hilt
 * With the SINGLEBUTTON compile directive you can configure your saber electronics
 * to use only a single button to interact with your board.
 */

// If your saber has only a single button to interact with the electronics, uncomment the next line
// in case you have 2 buttons (referred to as main and aux buttons) leave this line commented out

#define SINGLEBUTTON
 
#ifdef DIYINO_PRIME 
  #define MAIN_BUTTON     12
  #ifndef SINGLEBUTTON
    #define AUX_BUTTON   4
  #endif
#else if defined DIYINO_STARDUST_V2 or defined DIYINO_STARDUST_V3
  #define MAIN_BUTTON      12
  #ifndef SINGLEBUTTON
    #define AUX_BUTTON   11 // 2
  #endif
#endif

/*
 * BLADE TYPE
 *
 * RGB LED OR PIXELBLADE users:
 * Comment the following line will
 * disable and remove all LEDSTRINGS
 * blocks from compile
 *************************************/
//#define LEDSTRINGS
//#define STAR_LED
#define PIXELBLADE
//#define ADF_PIXIE_BLADE

/************************************/
[... 345 lines cut ...]
 * For daily use I recommend you comment LS_INFO
 * When you plug your device to USB uncomment LS_INFO !
 */
#define LS_LOOPLENGHT
#define LS_SERIAL  //enable serial communication using Wire library
#if defined LS_SERIAL
//#define LS_FSM
#define LS_INFO
//#define LS_DEBUG
#endif

#if defined LS_DEBUG
//#define LS_BUTTON_DEBUG
//#define LS_MOTION_DEBUG
//#define LS_MOTION_HEAVY_DEBUG
//#define LS_RELAUNCH_DEBUG
//#define LS_DEBUG_SLEEP
#endif

#if defined LS_MOTION_DEBUG
//#define LS_SWING_DEBUG
//#define LS_SWING_HEAVY_DEBUG
//#define LS_CLASH_DEBUG
//#define LS_CLASH_HEAVY_DEBUG
#endif`

Protonerd, 2020-05-02:
My first guess is that your switch is broken. Can you check this?

Rememberthecreed, 2020-05-02:
I did. It was part of the issue, the other was that I didn’t bridge all three rgb leads. I’ve got it working, thank you.

closed 2020-05-02 by Rememberthecreed

Protonerd, 2020-05-03:
Great, glad it works again!!!