Protonerd/FX-SaberOS #99: Hum and development

fadroid, 2020-05-02:
Hello,

I am having the issue where the him for the last sound font position does not work. I noticed a closed thread but did not see a resolution. I have done 3 separate builds, both single and dual button versions. Pretty much everything works as designed except the hum on the sound font in the last position. I have tried varying the number of sound fonts, and I always format before moving the files over. Did this ever get resolved? 

Also, is this project still in development?

Protonerd, 2020-05-03:
Interesting, I never had this issue in all these years! He many sound fonts do you attempt to use? I personally never went beyond 5. Is there a lower limit from which downwardsit starts working again?

fadroid, 2020-05-03:
I started with the 5 soundfonts that were included and tried as many as 10-12 but seem to have other issues past 7. I will attempt to go lower than 5 and see what happens.

normanbos, 2020-05-07:
I am having the same issue with all my builds! And I never found a solution to this.

Protonerd, 2020-05-07:
Did you try it with 5 or less Sound fonts? Since it is a mature and stable System, it is free of bugs. However the Firmware was qualified for a max number of 5 Sound fonts. Above that you are on unchartered terrain.

normanbos, 2020-05-07:
I did try it with less than 5 soundfonts, but it is always the last font that has the problem. Could the problem be cheap DFPlayer Mini modules?

fadroid, 2020-05-07:
Yes, I have also tried the original 5 that are provided on all the builds with the same result. I have not had the time yet to try less than 5.will try to get to it this weekend.

sgtglockcoma, 2020-05-20:
I have been having this issue as well but have found a solution that worked for me. I have a homebrew setup with 5 soundfonts, In the code I left the number of soundfonts at 5 but on my sd card I made a sixth soundfont file and filled it with a duplicate of my 5th soundfont. Still not sure why this is an issue but this seems to be a work around.

jbkuma, 2020-05-20:
It is strange.  I haven't been able to duplicate this.  At least we have a
temporary fix for those seeing this issue.
Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

fadroid, 2020-05-20:
That is a genius workaround! I just tried it and it works like a charm. I did 6 fonts on the system but loaded 7 on the card. Thank you so much!

Protonerd, 2020-05-24:
Great workaround, glad it works! I'm still not sure why it happens for you guys, as we in the development team never had this problem.

closed 2020-05-24 by Protonerd