Protonerd/FX-SaberOS #104: No sound - Speaker without amplifier or additional power?

jasonglisson, 2021-01-12:
I've been working on building a homebrew saber for a few days now. I have the blade igniting and the MPU is working, however, I have not been able to get any sound other than a buzzing sound from Dfplayer. 

Another project I'm working on I was able to get Dfplayer working just fine, however, this project I can't get a single sound from it. 

I've tried several Dfplayer modules. Same result.

I've tried several speakers. Same result.

And then I hit on this post about the exact same issue and it seem, perhaps, this is missing from the schematics. 
https://github.com/Protonerd/FX-SaberOS/issues/68

Can you confirm that you will not get any sound at all if you are prototyping with breadboards and the USB power supplied to the Arduino? From that thread, it sounded like you have to have external power for the speakers or an amplifier. 

I don't have a 5v power supply and I was trying to get this figured out before I spent money on batteries or anything else.

Any suggestions on how to get sound without the 5v adapter or additional power to the speakers just for prototyping? 

Excellent work on all of this!

jasonglisson, 2021-01-12:
I confirmed this one my own. I found some random adapter for something that was rated for 5v. So I just cut the end off it and wired it to my board and, sure enough, the speaker worked. 

I think this needs to be noted somewhere in the schematic or notes if someone is prototyping using their USB.

closed 2021-01-12 by jasonglisson

jbkuma, 2021-01-12:
It doesn't need 5v, but it needs sufficient current to operate (nothing in the install needs 5v) the amp must be directly powered from your source, preferably a charged lithium ion battery.

jasonglisson, 2021-01-12:
Thank you for weighing in on that. I've seen several people run into this issue just prototyping using the USB as the power source and that's definitely not going to work it seems.