Protonerd/FX-SaberOS #107: Sound problem with the Config Mode

NessKafetheYoungling, 2022-06-19:
Hi, I'm trying to make my first lightsaber with an arduino nano. 
I did all the wiring( except the part with the battery because I don't have it yet) and I have a problem with the config mode. 
When I make a long press on the auxiliary button , I have a weird sound and then I get in the config mode. Beside I don't have the sounds telling which setting I'm into (like soundfonts , main color etc). 

Any idea how to resolve that?

Druidjitsu, 2023-10-26:
DFPlayers with the JC chipset (which are everywhere and cause much trouble) are known to cause intermittent issues like these.  I myself experienced similar repetitive loops with my lightsaber sounds.
The Facebook Forum for FX-SaberOS recommend DFPlayers using either the YX5200 or YX5300 chipsets.  Replace your DFPlayer with JC, to ones with YX5200/5300.

rhansenne, 2024-02-16:
The latest version of the code contains improved support for clone DFPlayer modules. Activate DFPLAYER_CLONE in Config_HW.h to enable improved compatibility.

closed 2024-02-16 by rhansenne