Protonerd/FX-SaberOS #108: problem with arduino nano and mpu

pocotokomu, 2022-07-26:
when i try to calibrate the mpu i get:

Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano Every, ATMEGA328"

C:\Users\kacpe\Desktop\FX-SaberOS\tools\MPU6050_calibration\MPU6050_calibration.ino: In function 'void setup()':

MPU6050_calibration:76:2: error: 'TWBR' was not declared in this scope

  TWBR = 24; // 400kHz I2C clock (200kHz if CPU is 8MHz). Leonardo measured 250kHz.

  ^~~~

C:\Users\kacpe\Desktop\FX-SaberOS\tools\MPU6050_calibration\MPU6050_calibration.ino:76:2: note: suggested alternative: 'TWI0'

  TWBR = 24; // 400kHz I2C clock (200kHz if CPU is 8MHz). Leonardo measured 250kHz.

  ^~~~

  TWI0

Multiple libraries were found for "EEPROMex.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx

 Not used: C:\Program Files (x86)\Arduino\libraries\EEPROMEx

Multiple libraries were found for "I2Cdev.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\I2Cdev

 Not used: C:\Program Files (x86)\Arduino\libraries\I2Cdev

Multiple libraries were found for "Wire.h"

 Used: C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire

 Not used: C:\Program Files (x86)\Arduino\libraries\Wire

 Not used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\Wire

Multiple libraries were found for "MPU6050.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\MPU6050

 Not used: C:\Program Files (x86)\Arduino\libraries\MPU6050

exit status 1

'TWBR' was not declared in this scope

but when i comment out the twbr line i  get:

Arduino: 1.8.19 (Windows 10), Board: "Arduino Nano Every, ATMEGA328"

In file included from C:\Users\kacpe\Desktop\FX-SaberOS\libraries\I2Cdev\I2Cdev.h:86:0,

                 from C:\Users\kacpe\Desktop\FX-SaberOS\libraries\I2Cdev\I2Cdev.cpp:46:

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h: In static member function 'static int8_t I2Cdev::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*, uint16_t)':

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:63:12: note: candidate 1: size_t TwoWire::requestFrom(int, int)

     size_t requestFrom(int, int);

            ^~~~~~~~~~~

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:61:12: note: candidate 2: virtual size_t TwoWire::requestFrom(uint8_t, size_t)

     size_t requestFrom(uint8_t, size_t);

            ^~~~~~~~~~~

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h: In static member function 'static int8_t I2Cdev::readWords(uint8_t, uint8_t, uint8_t, uint16_t*, uint16_t)':

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:63:12: note: candidate 1: size_t TwoWire::requestFrom(int, int)

     size_t requestFrom(int, int);

            ^~~~~~~~~~~

C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire\src/Wire.h:61:12: note: candidate 2: virtual size_t TwoWire::requestFrom(uint8_t, size_t)

     size_t requestFrom(uint8_t, size_t);

            ^~~~~~~~~~~

In file included from c:\users\kacpe\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0,

                 from c:\users\kacpe\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\pgmspace.h:90,

                 from C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/api/String.h:31,

                 from C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/api/IPAddress.h:24,

                 from C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/api/ArduinoAPI.h:30,

                 from C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\cores\arduino/Arduino.h:23,

                 from C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.h:24,

                 from C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp:23:

C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp: In member function 'bool EEPROMClassEx::isReady()':

C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp:91:9: error: 'NVM_STATUS' was not declared in this scope

  return eeprom_is_ready();

         ^

C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp:91:9: note: suggested alternative: 'GET_STATUS'

C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp:91:9: error: 'NVM_NVMBUSY_bp' was not declared in this scope

  return eeprom_is_ready();

         ^

C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx\EEPROMex.cpp:91:9: note: suggested alternative: 'NVMCTRL_FBUSY_bp'

Multiple libraries were found for "I2Cdev.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\I2Cdev

 Not used: C:\Program Files (x86)\Arduino\libraries\I2Cdev

Multiple libraries were found for "Wire.h"

 Used: C:\Users\kacpe\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.7\libraries\Wire

 Not used: C:\Program Files (x86)\Arduino\libraries\Wire

 Not used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\Wire

Multiple libraries were found for "MPU6050.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\MPU6050

 Not used: C:\Program Files (x86)\Arduino\libraries\MPU6050

Multiple libraries were found for "EEPROMex.h"

 Used: C:\Users\kacpe\Desktop\FX-SaberOS\libraries\EEPROMEx

 Not used: C:\Program Files (x86)\Arduino\libraries\EEPROMEx

exit status 1

Error compiling for board Arduino Nano Every.

i have  tried using "DIYino_Testsketch_Pro.ino" from another issue thread but  it basically ends the same, and i don't know anything about coding

Protonerd, 2022-07-27:
Which board do you define in the Arduino IDE ?
BTW, the MPU calibration is not strictly speaking necessary, the saber code will work without as well.

pocotokomu, 2022-07-27:
i was trying to program on arduino nano every because i had it on hand, and had it defined in arduino IDE, but i read today somewhere that arduino nano and nano every are different to program so i am waiting on a normal nano and will try with it when it arrives

closed 2022-08-06 by pocotokomu