Protonerd/FX-SaberOS #109: Problem with the MP3-TF-16P sound

amaury14, 2022-07-29:
When the saber get power, the boot sound is heard, but some times it gets cut (and no more sound is heard - the player disconnects), and for sure when I start the saber (Main Button pressed), the leds get on, but the sound start and gets cut in the middle, the hum, never or almost never gets to be reproduced. Doing a DEBUG in the sound library, when that happens, I can see an error "DFPlayer: Unhandled sended message: " and some data, so I'm sure the Arduino Nano is sending the instruction, but the MP3-TF... gets disconnected for some reason. Maybe is sound files issues, or am preparing the SD card wrong, but I followed the instructions. Any help, please. Thank you in advanced and this is great work.

amaury14, 2022-07-29:
An using the sounds provided in this project and also used the Jakesoft converter, and added 120 seconds more to the original files, I had the same problem for the original files. I've tried both ways.

amaury14, 2022-07-31:
After some research (am not electronic, am developer). I found that the 5Volts provided by the Arduino Nano (connected to the PC, by USB) was not enough to start the lights and the sound at the same time, that's why the issue described above was happening. Solution use a po0wer source with the next voltages 7.4-8.2, supply to the LED strip no more than 5 volts, but the Arduino can work with all that power, using the VIN port (7-10V). Hope this helps somebody with the same issue. Cheers.

closed 2022-07-31 by amaury14

jbkuma, 2022-08-01:
Hi Amaury!

You want to power the Arduino via from a 3.7V battery via the 5V pin, and
the sound and lights need to be powered directly from the battery, not
through arduino/USB power.  You do not need to run on a double cell set up,
you also SHOULD NOT do that.

This is a common issue, and it is a user installation issue, not a design
issue.

I hope that helps!

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

amaury14, 2022-08-01:
Thank you for answer and for the info!!!!. I'll do it that way.