Protonerd/FX-SaberOS #110: sound problems

pocotokomu, 2022-08-06:
so i have finally gotten everything quite working, but when i retract the blade the retracion sound gets looped, it stops when i reignite the saber or go into config, but it's still there  the next time i retract it.

second problem i have is that in config mode some menu sounds get cut off or i.e. when i change through the sound fonts there is a beeping sound like it's instead of a name of the font.

fonts are default from the fxsaber os file and are added to the sd card like the wiki says, after one formating and adding files to the card the retraction loop stopped for one time, the next time the loop appeared again.

also i'm not able to go into silent mode, not sure when this happened but it was working a few times ago

Druidjitsu, 2023-10-26:
DFPlayers with the JC chipset (which are everywhere and cause much trouble) are known to cause intermittent issues like these.  I myself experienced similar repetitive loops with my lightsaber sounds.
The Facebook Forum for FX-SaberOS recommend DFPlayers using either the YX5200 or YX5300 chipsets.  Replace your DFPlayer with JC, to ones with YX5200/5300.

rhansenne, 2024-02-02:
I had the same issue with the MH2024K-24SS (clone) version of DFPlayer and had to make some minor code changes to correct this. I created the following pull request, which should fix the issue when uncommenting "DFPLAYER_CLONE" in Config_HW: https://github.com/Protonerd/FX-SaberOS/pull/113

closed 2024-02-16 by rhansenne