Protonerd/FX-SaberOS #111: Exceeds available space on Nano with DIYINO_PRIME and PIXELBLADE

bombayjesus, 2023-02-13:
Hallo! I absolutely love the level of detail that has gone into this repository - really, really appreciate it!

So I managed to rig up the specification with an Arduino Nano and a 97 x 2 LED long Pixelblade. If I change the HW configuation to a DIYINO_PRIME instead of DIYINO_STARDUST_V3, the end program becomes too big to upload to the Nano :(

It gets even bigger once I change it to the Pixelblade further down the file.

However, if I leave it to DIYINO_STARDUST_V3, it boots up as expected but I'm not sure what might be failing because I followed the instructions for Homebrew DIYINO_PRIME.

Has anyone had this happen? Would really appreciate any help!

Also, do let me know if I should copy-paste any of the HW/SW, or other, configuration files for reference.

Thank you again everyone who's contributed to this brilliant repo!

bombayjesus, 2023-02-13:
I have found a way to get it to compile! If you disable LS_INFO, it works out. If it remains enabled, it goes over the memory limit. I assume LS_INFO is to help debug? Is there any way to compile and upload WITH LS_INFO?

rhansenne, 2024-02-14:
The program storage space used will depend on the features enabled in Config_HW and Config_SW. Comment out any features  (e.g. jukebox, deep sleep, battery check, pixel accent, gravity color,...) you don't plan on using.
Also, when not debugging, comment out LS_SERIAL (which will also disable LS_INFO and LS_DEBUG) to gain space.