Protonerd/FX-SaberOS #112: Df player freezes and resets

DootSlayer99, 2024-02-02:
Well i did everything correctly and sounds are working while swinging however it freezes and resets back to boot around every 1 or 2 minutes. It also loops the poweroff sound and the config sounds like main color and swing sensitivity. How can i fix this?

DootSlayer99, 2024-02-03:
It works, now it doesnt loop! Thanks a lot! I am still trying to figure out why it resets for no reason. Can it be because of fake df player?

rhansenne, 2024-02-16:
I can't reproduce the reset behavior. This sounds like a possible loose connection somewhere. Did you manage to track down the cause? The latest version of the code (in master) contains some further compatibility improvements for clone DFPlayer modules (such as a configurable DFPLAYER_OPERATING_DELAY in Config_HW), which may help.

Protonerd, 2024-02-29:
Strangely for me it resets every few secounds and will never enter the main loop. The last message I get is about the DFPlayer being initialized and then the watchdog triggers. If watchdog is not enabled, the board just freezes. I used one of my known-working sabers for this trial, which worked for years, up to the point yesterday when I uploaded the new code. Funnily even though it's my board design and a code mostly written by me, for the first time I have no idea what I could still try.

rhansenne, 2024-02-29:
@Protonerd does that saber still work correctly with the older [v1.4](https://github.com/Protonerd/FX-SaberOS/tree/v1.4) version of the code? If so, we can debug which of the recent changes may be causing the issues.

DootSlayer99, 2024-02-29:
Well after some testing i actually found out why it resets! The code works for me but the reason why it resets for me is because of the custom font i created on my own. I forgot to use the soundfont converter and when i used it the resetting issue was gone. Although the fake df player still reset after a couple seconds even if i use the sample soundfonts given in the master zip. Original work well though. @Protonerd perhaps there is an issue with the soundfonts you used?

rhansenne, 2024-03-04:
I'm currently using Arduino IDE v2.2.2 with AVR package version 1.8.6.

Protonerd, 2024-03-13:
I was able to recover my saber with the last release before the recent ones. I think the problem was the AVR package, the newer ones do not have an option for the bootloader my DIYino boards are loaded with. So I went back to AVR 1.6.17 where I still found the board type Arduino/Genuino UNO, which hides the optiboot BL, which has slightly more memory left for code. With that one I was able to reload the code and now the saber works fine. Now I'm going to be more cautious and try the new features with a bare Nano board.
I'm not sure why but earlier I did not have to take care about certain libraries staying in the cache and causing compile troubles but now most of the time I need to purge the cache every time I make changes to libraries. Crazy...

closed 2024-04-19 by rhansenne